mavlink/qgroundcontrol #12953: QGC v5.0 (RC2 / RC3) issue with MavLink camera control protocol UI

nrplyx, 2025-05-31:
Testing QGC v5.0 (tested RC2 and RC3) with our new camera controller (TAG-E) and have issues with camera control UI (loaded based on XML file.
After launch of QGC is UI rendered correctly 

[image]

However, after any user interaction with UI, QGC removes all camera controls. Error in console is:

[image]

And leading to this line:
https://github.com/mavlink/qgroundcontrol/blob/bda50eb4f3fac456f61d687f1d97f89f13c206c4/src/FlightMap/Widgets/PhotoVideoControl.qml#L506
Which is about slider for interval shooting configiuration.

------
In QGC v4.4.5 all works correctly and stable.

DonLakeFlyer, 2025-06-06:
> QGC removes all camera controls

Not exactly sure what you mean by that? You change something in the dialog and the dialog disappears? You change something in the dialog, close it and then some part of the UI is no longer visible? Can you provide more detail?

DonLakeFlyer, 2025-06-06:
Hmm, the binding loop error doesn't show up for me

DonLakeFlyer, 2025-06-06:
Once my Console fix finishes building you should be able to grab a new build from GitHub Actions. Use that build and let me know what Console says when you repro this.

DonLakeFlyer, 2025-06-10:
I added another fix to #12998. Try that...

PR #12998 (2025-06-11): Fix parameter name conversion from mavlink
* Previous code was failing with new 6.8. Switch to same code which is used in ParameterManager.
* Fix #12953 (hopefully!)