mavlink/qgroundcontrol #12956: Daily QGC 5.0 for android UI fails to give scroll control

TimothyGold, 2025-06-02:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
First open of android app you can go into the application settings and scroll through the tabs and options. Next open the scroll bars are gone and you can no longer scroll through the tabs or the settings. 

**Current Behavior**
Describe what actually happened here.

**Steps To Reproduce**
1. Install
2. Close
3. Reopen 

**Additional Details**
Add any extra info (errors, warnings, etc.) here.

### System Information

Running android version 15

igoutta, 2025-06-02:
I can condone this. Doesn't happen in all settings, but it happens in some like Genral Settings. Besides the UI size doesn't match with the phone. The default overflows the phone.

DonLakeFlyer, 2025-06-06:
Also just double-checking: You realize the scroll bars are just visual indicators which disappear after a fixed time? You can still scroll when there are not visible.

TimothyGold, 2025-06-06:
Visuals are one thing.  Actually not being able to scroll is another. I cannot scroll in any way is some pages. Works sometimes then stops.

TimothyGold, 2025-06-06:
I just tried it again. Opened general application settings and the scroll bars showed up. I can use my fingers to scroll up down in sides of the page. Used the shortcut to close the app. Restarted and opened application settings...  can no longer scroll in any way....

[]()[image]

DonLakeFlyer, 2025-06-06:
What case is this image from? I'm assuming the case where you can scroll since the scroll indicator is showing.

TimothyGold, 2025-06-06:
No, this screenshot was when I could not scroll. After finger scroll failed.

TimothyGold, 2025-06-06:
I am willing to test. 

When the scroll fails I look for any handle or scroll bar to grab to move the page. But nothing shows up. The general finger scroll is also gone and does not work either.

DonLakeFlyer, 2025-06-09:
@TimothyGold The build artifacts from #12978 should be available now for you to test with. There is console logging I want to capture. So uninstall, run it, General Settings, capture console log output. Run again, General Settings (hopefully fails to scroll), capture log output.

TimothyGold, 2025-06-10:
I uninstalled and installed the new nightly build. When started the initial dialogs could not even be seen. I could not scroll to set anything just could click OK to continue. Went to general settings to enable "capture console log output but no matter what I do cannot scroll to the bottom to find this setting. Closed,  reset phone to try to scroll. Nothing allowed me to scroll. So no log can be captured. 

[image]
[image]

DonLakeFlyer, 2025-06-10:
Oh, crap. I'm just realizing you can't use an APK from CI since it's not signed. Duh! Just a sec...

DonLakeFlyer, 2025-06-10:
You need to use the builds created from the pull request CI run not the nightly build. If you go to GitHub Actions.

Expand the checks section like this:
[image]

Then pick the Release build for the OS you are on. For example for Windows it would be "Windows / build (Release, x64)". Click Summary on the page you end up on after clicking that. At the bottom of the page there is an Artifacts section:

[image]

Download the installer/build from there and run that.

There is no settings to capture console output in General. Go to Application Settings / Console and the output is there with the option to "Save App Log".

DonLakeFlyer, 2025-06-11:
@TimothyGold Try a new daily build. I pushed some window sizing changes.

TimothyGold, 2025-06-11:
Just downloaded the daily build. When it first starts I can get through the first dialogs and scroll fine. I can also open application settings and scroll works great. I then use the exit and close option in the main menu and restart the app. Scroll options are again not working in the application settings menu. 

[image]

TimothyGold, 2025-06-12:
Not in a place to connect to flight controller. When i go to the plan flight screen this is what it looks like. Again no scroll options here either. 

[image]

TimothyGold, 2025-06-12:
[image]

DonLakeFlyer, 2025-06-12:
Ok, can you do this:
* Uninstall QGC
* Install QGC from here: https://qgroundcontrol.s3.us-west-2.amazonaws.com/DonLakeFlyer/android-build-QGroundControl-debug.apk
* Boot QGC
* Go to App Settings / Console. You should seem something like this:
[image]
* Capture a screen shot
* Close QGC
* Boot QGC
* * Go to App Settings / Console. Capture another screen shot
* Post both screen shots here

TimothyGold, 2025-06-13:
I uninstalled and installed from your link. So far so good. I can scroll in all screens.