mavlink/qgroundcontrol #12992: Missing baudrates in Comm Links

riosnav, 2025-06-09:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
I am configuring QGroundControl on a new laptop with the latest version from the website (5.0.2) and want to add a Comm Link with Baud Rate 500 000 as I have in my older device running 4.4.4.

**Current Behavior**
When adding a new Comm Link, it is not possible to select baudrates higher that 256000 in the dropdown menu.

**Steps To Reproduce**
1. Application Settings
2. Comm Links
3. Add New Link
4. Baud Rate dropdown menu

**Additional Details**
Maybe I am missing something. In case this is indeed a new limitation, can I circumvent the menu to input the value editing some config file?

### System Information

v5.2.0 64 bit
Windows 11

riosnav, 2025-06-09:
I have tried manually editing QGroundControl.ini to change the baudrate, but upon restart of QGroundControl it defaults to the lowest baudrate 110, at least graphically.

HTRamsey, 2025-06-09:
The baud rate options for windows comes from [here](https://github.com/qt/qtserialport/blob/61573f5054fceb6b0a79c000f88479beeb14a385/src/serialport/qserialport_win.cpp#L775) which is apparently obtained from [WinBase.h](https://github.com/tpn/winsdk-10/blob/9b69fd26ac0c7d0b83d378dba01080e93349c2ed/Include/10.0.10240.0/um/WinBase.h#L357). We used to have them hard coded in the [SerialLink file](https://github.com/mavlink/qgroundcontrol/blob/08b77a684b306560061db9d59e8b33f6cd3353bf/src/comm/SerialLink.cc#L412). Would a custom/manual entry work for you?

riosnav, 2025-06-09:
> The baud rate options for windows comes from [Qt's qserialport_win.cpp](https://github.com/qt/qtserialport/blob/61573f5054fceb6b0a79c000f88479beeb14a385/src/serialport/qserialport_win.cpp#L775) which is apparently obtained from [WinBase.h](https://github.com/tpn/winsdk-10/blob/9b69fd26ac0c7d0b83d378dba01080e93349c2ed/Include/10.0.10240.0/um/WinBase.h#L357). We used to have them hard coded in the [SerialLink file](https://github.com/mavlink/qgroundcontrol/blob/08b77a684b306560061db9d59e8b33f6cd3353bf/src/comm/SerialLink.cc#L412). Would a custom/manual entry work for you?

I think it would work fine, yes.

HTRamsey, 2025-06-09:
We can, but personally I think it's better to obtain the baud rates that the platform libraries actually support, then allow a user to force a custom baud if they want to try that. Even the different Android drivers support different baud rates so it's difficult to hard code an accurate mapping

PR #13017 (2025-06-17): Comms: Add Default List of Serial Baud Rates
Closes #12992. Would like to add custom entry box too.