mavlink/qgroundcontrol #12995: RTSP Streaming Latency in Daily QGC 5.0 for Android

Eureak-zon, 2025-06-09:
### Your Question

There is significant RTSP streaming latency when running Daily QGC 5.0 on an Android tablet. Disabling frame synchronization greatly reduces the latency, but introduces noticeable screen tearing.

Interestingly, when using an emulator with a direct wired network connection, RTSP streaming performs with no latency issues at all.

Is there any recommended solution or best practice to reduce RTSP latency on Android devices while maintaining smooth video playback?

HTRamsey, 2025-06-09:
I'm going to get you to try #12918 once it finishes building

Eureak-zon, 2025-06-09:
> can you try [#12918](https://github.com/mavlink/qgroundcontrol/pull/12918) once it finishes building?

yes,sure!

HTRamsey, 2025-06-10:
It's got some kinda error, let me fix it and I'll give you a link to download

Eureak-zon, 2025-08-01:
The latest version of QGC still has this issue. The previous modification was effective.
[Commit 7684968](https://github.com/mavlink/qgroundcontrol/commit/76849686c598f57e6963dc14aead60bf684642c4)

closed 2025-08-01 by Eureak-zon