mavlink/qgroundcontrol #13068: Building without GStreamer breaks MAVLink live plotting

aWZHY0yQH81uOYvH, 2025-06-22:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
MAVLink plots display data.

**Current Behavior**
If built without GStreamer, the traces on the plots inexplicably disappear. The plot axes appear and update with appropriate values, but the actual line for the data is not present.

**Steps To Reproduce**
1. Build with `-DQGC_ENABLE_GST_VIDEOSTREAMING=NO`
2. Go to MAVLink inspector
3. Select an item to plot

### System Information

* QGC version: 5.0.3
* QGC build: from source
* Operating system: macOS 15.5 (x64)
* Flight controller: Pixhawk 6X
* Autopilot software: ArduPlane 4.6.0

### Log Files and Screenshots

My test commit: https://github.com/aWZHY0yQH81uOYvH/qgroundcontrol/commit/c9438d86cdbe40ceab3a058709b73df01fab3806
Test CI build: https://github.com/aWZHY0yQH81uOYvH/qgroundcontrol/actions/runs/15802066126

[image]

DonLakeFlyer, 2025-06-23:
Hard to believe, but this is real! Poked at it a bit with no luck so far.

DonLakeFlyer, 2025-06-25:
Some sort of OpenGL problem. If you change this line to `false it works:
https://github.com/mavlink/qgroundcontrol/blob/master/src/QmlControls/MAVLinkChart.qml#L34

DonLakeFlyer, 2025-06-25:
I'm assuming GStreamer on Mac ends up bring along some OpenGL thing with it that Charts needs. @HTRamsey Anything ringing a bell here with you?

DonLakeFlyer, 2025-06-25:
Last ditch hack would be: `serie.useOpenGL = QGroundControl.videoManager.gstreamerEnabled`

HTRamsey, 2025-06-25:
Oh interesting, that's a strong possibility. Could link to Qt6OpenGL regardless of GStreamer

DonLakeFlyer, 2025-07-01:
Holden:"I'd try linking to Qt6OpenGL all the time regardless of GStreamer and that should fix it"