mavlink/qgroundcontrol #13074: No map overlay when sending MAVLink OBSTACLE_DISTANCE

Mahmood-Hussain, 2025-06-23:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
See the object map overlay on the QGroundControl map

**Current Behavior**
Can not see the the objects overlay getting error qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments

**Steps To Reproduce**
1. Connect companion computer (micro-controller) via TELEM2
2. Enable MAV_1_CONFIG (set it to TELEM2) reboot FC
3. Enable MAV_1_FORWARD
4. COM_OBS_AVOID Enabled
5. CP_DIST (POSITIVE) 5m
6. Go to analyze tools to verify Heartbeat and OBSTACLE_DISTANCE is received (YES)
7. Check obstacle map overlay on qgc: (NO) -> ERROR in the console 

**Additional Details**
I am using Teensy4.1 micro controller and MAVLink library https://github.com/okalachev/mavlink-arduino (okalachev/MAVLink@^2.0.16) sending OBSTACLE DISTANCE and heartbeart I can see both in analyze tools but then I get error when I try to see the map overlay. I tried the same code and way on Ardupilot, MissionPlanner shows that radar like obstacle map successfully.

For the reference here is how I send both
```
// Send Heartbeat message
void MAVLinkBridge::sendHeartbeat() {
    uint32_t currentTime = millis();

    // Only send heartbeat if interval has elapsed
    if (currentTime - _lastHeartbeatSendMs >= HEARTBEAT_INTERVAL_MS) {
        mavlink_message_t msg;
        uint8_t buf[MAVLINK_MAX_PACKET_LEN]; // Use MAVLINK_MAX_PACKET_LEN for safety

        // Common heartbeat fields
        uint8_t base_mode = MAV_MODE_FLAG_CUSTOM_MODE_ENABLED; // Indicate we are in some custom mode
        uint32_t custom_mode = 0; // Placeholder
        uint8_t system_status = MAV_STATE_ACTIVE; // Indicate component is active

        mavlink_msg_heartbeat_pack(
            TEENSY_SYS_ID,                   // System ID
            TEENSY_COMP_ID,                  // Component ID
            &msg,                            // Message object
            MAV_TYPE_ONBOARD_CONTROLLER,     // Type: Use a suitable type (GIMBAL, PERIPHERAL, etc.)
            MAV_AUTOPILOT_INVALID,           // Autopilot type (indicates generic companion)
            0,
            custom_mode,
            system_status
        );

        uint16_t len = mavlink_msg_to_send_buffer(buf, &msg);
        _serialPort.write(buf, len);

        _lastHeartbeatSendMs = currentTime;
        // DEBUG_SERIAL.println("Sent Heartbeat"); // Avoid frequent debug prints
    }
}

// Send OBSTACLE_DISTANCE message
void MAVLinkBridge::sendObstacleDistance(uint64_t time_usec, uint16_t min_distance_cm, uint16_t max_distance_cm, float increment_f, float angle_offset, const uint16_t* distances) {
    mavlink_message_t msg;
    uint8_t buf[MAVLINK_MAX_PACKET_LEN]; // Use MAVLINK_MAX_PACKET_LEN for safety

    // Ensure distances array has 72 elements (MAVLink spec)
    static_assert(MAVLINK_OBSTACLE_DISTANCE_BINS == 72, "MAVLink OBSTACLE_DISTANCE requires 72 bins");

    // Pack the OBSTACLE_DISTANCE message
    mavlink_msg_obstacle_distance_pack(
        TEENSY_SYS_ID,                 // system_id
        TEENSY_COMP_ID,                // component_id
        &msg,                          // msg
        time_usec,                     // time_usec: Timestamp (use vehicle's or your own)
        MAV_DISTANCE_SENSOR_LASER,     // sensor_type: Laser based
        distances,                     // distances: Array of uint16_t distances in cm (exactly 72 elements)
        0,                             // increment: Angular increment in degrees (deprecated, use increment_f) - Set to 0
        min_distance_cm,               // min_distance: Minimum distance sensor can measure (cm)
        max_distance_cm,               // max_distance: Maximum distance sensor can measure (cm)
        increment_f,                   // increment_f: Angular increment between measures (float, degrees). Should be 360/72=5.0 for a full scan.
        angle_offset,                  // angle_offset: Angle of the minimum angle measure (first element in distances array) relative to sensor's frame_of_reference. With BODY_FRD frame and angle 0 being forward, this should be 0.0.
        MAV_FRAME_BODY_FRD             // frame: Coordinate frame of the sensor data. BODY_FRD is standard for sensor mounted rigidly to vehicle.
    );

    uint16_t len = mavlink_msg_to_send_buffer(buf, &msg);
    _serialPort.write(buf, len);

    // DEBUG_SERIAL.println("Sent OBSTACLE_DISTANCE"); // Avoid frequent debug prints
}

```

### System Information

QGC Version: 5.0.3, 4.4.4 
QGC build: daily, stable
Operating System:  Windows 11 PRO, Ubuntu 20.04
Flight Controller: CubePilot Cube Orange Plus
Autopilot Software (with version): PX4 1.15.4

### Log Files and Screenshots

./QGroundControl.AppImage 
Settings location "/home/e300/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/e300/Documents/QGroundControl/Logs"
Map Cache in: "/home/e300/.cache/QGCMapCache300" / "qgcMapCache.db"
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
"v4.4.4"
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
ComponentInformationTranslationLog: Locale  "en_US"  not found in json
ComponentInformationTranslationLog: Locale  "en_US"  not found in json
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters cache match failed /home/e300/.config/QGroundControl.org/ParamCache/1_1.v2
FirmwarePluginLog: Unable to parse version info from file "https://api.github.com/repos/PX4/Firmware/releases"
QCoreApplication::postEvent: Unexpected null receiver
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml:42: Error: createRadialGradient(): Incorrect arguments

[image]

DonLakeFlyer, 2025-06-23:
I tried PX4 SITL with LIDAR and it seems to work correctly there. Two options to debug this:
* If you know how to build QGC it would be much easier to have you debug it. You can `console.log` the values going into `createRadialGradient` to see why it doesn't work.
* Create a telemetry log for me which shows the issue and I might be able to hack something to replay that and debug it myself.

Mahmood-Hussain, 2025-07-31:
sorry for the delayed response. I was busy with other stuff. I will give it a try may be next week.

DonLakeFlyer, 2025-09-01:
I'll re-open when I et more info

closed 2025-09-01 by DonLakeFlyer