mavlink/qgroundcontrol #13084: The RTSP video is not displayed and needs to be forwarded by VLC before it can be displayed

mbwgl, 2025-06-25:
### Your Question

Now I can open the RTSP video stream using the. exe file in QGroundControl v5.0- Release Candidate 1, but using V4.4.5, V4.4.4, V4.4.3, V4.3, etc. does not display the RTSP video stream. At this point, after forwarding the RTSP through VLC, I can use other versions of QGC to display the video. Can you tell me what's going on? Is it due to a version issue with Gstreamer? Or is it due to the format configuration issue of streaming? thank you!

mbwgl, 2025-06-25:
The Android version can display RTSP videos using V3.5.6, but cannot display videos using V4.2.0

HTRamsey, 2025-06-26:
Am I understanding correctly that v5.0 is working but the older ones are not?

mbwgl, 2025-06-27:
> Am I understanding correctly that v5.0 is working but the older ones are not?我是否正确理解 v5.0 可以工作，但旧版本没有？

After testing, I found that the PC version 5.0 can work, but V4 cannot. In addition, Android V5.0 and V4 are also not working, but V3.5.6 can work. I'm not very familiar with this aspect, can you tell me how to solve it? Is there a convenient and fast way to roll back the RTSP video streaming from Android version V4.3 to V3.5.6? I am very, very grateful that you can help me solve the problem

mbwgl, 2025-06-27:
> Am I understanding correctly that v5.0 is working but the older ones are not?我是否正确理解 v5.0 可以工作，但旧版本没有？

I found that the latest version of Windows V5.0 can display videos by default, but when the Video decode priority is switched to "Force software decoder", the video cannot be displayed. The log is shown in the attachment (I found that Android V5.0 also prompts this type of error). However, setting the Windows version to other modes can display videos, but the "Force software decoder" and "Force Vulkan decoder" on the Android version both fail to display.

[QGC V5.0 5.txt](https://github.com/user-attachments/files/20936898/QGC.V5.0.5.txt)