mavlink/qgroundcontrol #13095: Custom flight modes not shown

fsoest, 2025-06-26:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

Following up on #12603 we implemented a custom flight mode on our hardware. However, the mode does not show up in the mode selector in QGC. It is shown when using commander status as an External Mode. 

**Expected Behavior**
Custom Mode is shown in QGC mode list

**Current Behavior**
Mode is not shown

**Steps To Reproduce**
1. Install ROS2 + PX4, MicroXRCEAgent
2. Install PX4 ROS 2 Control Interface following [1]
3. Start QGC
4. ros2 run example_mode_manual_cpp example_mode_manual

**Additional Details**
When enabling the loggers as in the previous issue, the mode is never shown anywhere.

### System Information

- Operating System: Ubuntu 22.04.5, Windows 11
- QGC Version: 5 RC 5
- Flight Controller: CubePilot Cube Orange
- Autopilot (with version): PX4 1.15.4

DonLakeFlyer, 2025-06-26:
So I'm trying to install ROS2 so I can test this myself. But I get here:
```
parallels@parallels:~/Downloads$ pip install --user -U empy==3.3.4 pyros-genmsg setuptools
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

Not quite sure how to work around this?

DonLakeFlyer, 2025-06-26:
And then I run the ros2 command in the virtual environment as well?

DonLakeFlyer, 2025-06-27:
That just moved me on to one after another problem getting ROS2 setup on 24.04 arm. Giving up after spending too much time on that. I'm going to convert Mocklink to Standard modes support so I can test with that.

DonLakeFlyer, 2025-06-27:
Ok, I can repro this now with my new MockLink implementation

DonLakeFlyer, 2025-06-30:
If you turn on VehicleLog you might get more insight into what went wrong

fsoest, 2025-07-02:
I have turned on VehicleLog and tested on a 6X-RT as well, does not work on either.

[6x-rt.txt](https://github.com/user-attachments/files/21014891/6x-rt.txt)
[cubeorange.txt](https://github.com/user-attachments/files/21014892/cubeorange.txt)

DonLakeFlyer, 2025-07-03:
Does 4.4 work on the same hardware?

fsoest, 2025-07-03:
I will take a look, but was custom mode support not only added after 4.4?