mavlink/qgroundcontrol #13118: QGC v4.2.4 - v5.0.4 - Video quality issues with SIYI MK32 + R1M or A2 mini camera

feherbalint1996, 2025-07-01:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

Hi,

I’m using a SIYI MK32 controller with an R1M camera, and I’ve noticed video stream issues in QGroundControl versions newer than v4.2.4.
Unfortunately, I’m not an expert, so I can only describe the problem itself, not the solution.

Here are my observations:
**SIYI QGC and QGC v4.2.4:**
Video stream is good: stable, smooth, generally usable (though not quite as good as in the SIYI FPV app, but still acceptable).

**QGC v4.3 – v4.4.5:**
• When starting the app for the first time, the video quality seems about the same as in v4.2.4.
• However, after restarting the application (while keeping the remote controller and air unit powered on and connected), the video quality becomes noticeably worse and starts breaking up.
• Restarting the air unit or the remote controller makes the video good again.
• This issue does not occur in v4.2.4.

**QGC v5.0.0 – v5.0.1:**
Unfortunately, these versions cannot be installed on the SIYI MK32 controller, so I wasn’t able to test them.

**QGC v5.0.3 – v5.0.4:**
The video quality became extremely poor: almost nothing is recognizable.
• Many times, the entire video turns completely green.
• When it does show an image, there is huge delay and very bad quality.
• Restarting the air unit or the remote controller does not help.
• It also doesn’t help to start QGC first and then connect, or connect first and then start QGC.
• Changing the decoding type didn’t solve the problem (same as in earlier versions).

**The SIYI A2 Mini camera shows exactly the same issue as described above.
In all cases, I was using the latest available firmware on both the camera and the controller.**
**I also tested with another MK32 remote controller, another air unit, and another R1M camera, and observed exactly the same issues.**

If it would help, I’m happy to provide logs, screenshots, or further testing details.
Thank you 

### System Information

QGC v4.2.4 - QGC v5.0.4
Build: daily, stable
Android 9

DonLakeFlyer, 2025-07-03:
Are you using h265, try switching to h264.

feherbalint1996, 2025-07-04:
> Are you using h265, try switching to h264.

I don't know if it's possible to change the encoding type in the SIYI system.
I found firmware for the R1M camera labeled "H264"
(https://drive.google.com/drive/folders/1SdZlwZ57gVOrYMF4_nDggS-tDwIc8v40)
Unfortunately, it ended up bricking the camera. :(

rutger-seascape, 2026-04-15:
Hi @feherbalint1996 ,

Sorry to hear about your camera. It is possible to change the encoding type through the SIYI SDK. It's function "0x21: Set Camera Encoding Parameters". It's working on the SIYI A8. I contributed Python bindings for this function in [a Python implementation of the SDK](https://github.com/mzahana/siyi_sdk). The relevant test is [here](https://github.com/mzahana/siyi_sdk/blob/main/tests/test_set_encoding_param.py).