mavlink/qgroundcontrol #13147: Qgc5.0 version in FlightDisplayViewVideo. QML mouse events can not run

wang5d, 2025-07-09:
### Feature Description

Qgc5.0 version in FlightDisplayViewVideo. QML inside a mouse events, how to add, can response in version 4.0, cannot run on version 5.0, why

DonLakeFlyer, 2025-07-09:
Something else must be pulling in the left mouse button. You'll have to look in the Qml to find it.

closed 2025-07-09 by DonLakeFlyer

wang5d, 2025-07-10:
gpt says that GstGLQt6VideoItem overwrites the rendering layer and intercepts all mouse events by default, including left mouse buttons. I don't know if this is the reason

wang5d, 2025-07-10:
> Something else must be pulling in the left mouse button. You'll have to look in the Qml to find it.
gpt says that GstGLQt6VideoItem overwrites the rendering layer and intercepts all mouse events by default, including left mouse buttons. I don't know if this is the reason