mavlink/qgroundcontrol #13174: Wrong battery voltage

dollop80, 2025-07-14:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
The battery widget popup and quick view shold show battery voltage in volts

**Current Behavior**
The battery voltage shows illogical values inspite of the fact, the mavlink inspector shows correct values in milivolts

**Steps To Reproduce**
1. Power on a drone
2. Open QGC
3. Observe the battery viltage value

**Additional Details**
The QGC 4.4.4 shows correct voltage of the same vehicle

### System Information

QGC Version: [e.g. 5.0.6]
QGC build: [stable]
Operating System: [Windows 10 ]
Flight Controller: [VOXL 2]
Autopilot Software: [PX4 1.14]

### Log Files and Screenshots

[image]

DonLakeFlyer, 2025-07-27:
I need to see all the values from id to fault_bitmask. And The window needs to be wide enough such that no values are being clipped out.

dollop80, 2025-07-28:
No problem. This is with v5.0.6
[image]

Another fact. The QGC v5.0.6 dispalys correct voltage for first 0.2-0.5 seconds as the interface renders

This is with v4.4.4
[image]