mavlink/qgroundcontrol #13248: How does protocol version checking work?

hamishwillee, 2025-07-31:
### Your Question

I know broadly from https://mavlink.io/en/guide/mavlink_version.html#version_handshaking but I want to confirm that you do.

Specifically, 
1. It [appears you use](https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L2605) `MAV_CMD_REQUEST_PROTOCOL_VERSION` instead of `MAV_CMD_REQUEST_MESSAGE` with the id 300 for requesting the version. Right? 
   - So I can't remove PX4 support for the first message?
2. This isn't supported by ArduPilot right?
3. What do you do with the values? Current PX4 is at 200 max, but current mavlink is 203. In https://github.com/PX4/PX4-Autopilot/pull/25343 I am updating to 203. 
   - What will happen in QGC with this value. Will you reject the connection, or can you support 203?

mrpollo, 2025-08-05:
@DonLakeFlyer @HTRamsey can you help out here please?

hamishwillee, 2025-08-06:
Thanks very much @HTRamsey !!

That is "somewhat" good to hear. Feel free to remove any evidence of MAV_CMD_REQUEST_PROTOCOL_VERSION - at some point that will enable removal from the XML. Ditto any other specific message requestors that have been replaced by MAV_CMD_REQUEST_MESSAGE (all of them),. 

As discussed in https://github.com/mavlink/mavlink/pull/2318#issuecomment-3138399934 I'm not sure PROTOCOL_VERSION achieves what it is supposed to, mostly because MAVLink doesn't actually do more than token versioning of the standard. If we did we'd have a host of other problems in managing versions.

What will probably happen is that we will move it from minimal.xml to common.xml, which should not change anything. 

[The reason being that it isn't minimal if you don't actually need it to have a valid mavlink system, and the fact that ardupilot don't use it means you'd don't.
I'd push for it strongly if it was useful, but it isn't.]