mavlink/qgroundcontrol #13250: Cannot share telemetry files after installing QGC 5.0 for Android

einglez, 2025-07-31:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

I have recently installed the latest QGC 5.0 stable version on a Moto Neo 50 running Android 15, but I cannot share telemetry (*.tlog) files. I tried several exploreres with  no success, USB over pc, update do drive and adb.

I can share files at the same folder (android/data) of other apps (e.g Mission Planner), but not with QGC 5.0 stable.

### System Information

Latest 5.0 stable

HTRamsey, 2025-08-01:
What do you mean by share

einglez, 2025-08-01:
Send it over Gmail, WhatsApp ...

HTRamsey, 2025-08-22:
Can you check if the permissions of that file are different than the rest

einglez, 2025-08-22:
Info attached

[image]

HTRamsey, 2025-08-26:
Should be, it was definitely a read permission issue because I had the issue too

PR #13330 (2025-08-26): Comms: Fix TLOG Permissions
Fixes #13250
Uses a QSaveFile and sets file permissions
Also need to look into Android Permissions changes...