mavlink/qgroundcontrol #13283: Custom/external flight mode not displayed

nlsxp, 2025-08-11:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
I am using the px4 ros2 interface lib with ROS2 Jazzy to register an external flight mode in px4. It is expected that the registered flight mode is displayed in the available flight modes and the corresponding MAVLink messages should appear in the MAVLink inspector. This behaviour is true for SITL (for the most recent Q-GroundControl version and v4.4.0 rc-1). 

**Current Behavior**
When the actual flight controller (pixhawk 6XRT) is connected via USB serial connection the flight modes do not show up. Neither in the available modes nor the corresponding messages. The ros node however still registers the flight mode correctly. When entering the NuttShell(NSH) and excecuting the command ``commander status`` I see the following feedback, confirming that the flightmode is registered: 
``
nsh> commander status
INFO  [commander] Disarmed
INFO  [commander] navigation mode: Hold
INFO  [commander] user intended navigation mode: Hold
INFO  [commander] in failsafe: no
INFO  [commander] External Mode 5: nav_state: 27, name: Custom Mode
``

**Steps To Reproduce**
1. Follow the [PX4 ROS2 User Guide](https://docs.px4.io/main/en/ros2/user_guide.html) to setup the px4 ros2 interface library. 
2. Start the one of the examples in ``px4-ros2-interface-lib/examples/cpp/modes/`` 
3. connect ros to SITL or flightcontroller and QGC to see the bahviour above. 

### System Information

* QGC Version: v4.4.0 rc-1 / 5.0.6
* QGC build: Linux version on WSL and Windows download for each of the versions.
* Flight Controller: Pixhawk 6XRT
* Autopilot Software: [PX4-Autopilot v1.15.4](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.15.4)

nlsxp, 2025-08-12:
I ran QGC with StandartModesLog enabled. These lines were returned referencing the StandardModesLog:
    ``16.945 - debug: Requesting available modes (StandardModesLog:unknown:0)``
    ``17.964 - debug: Failed to retrieve available modes - REQUEST_MESSAGE:MAV_RESULT 4 (StandardModesLog:unknown:0)``

The full log is attached. No new logging messages appeared, once I started the node. I ran this on the newest version (v5.0.6 64 bit)

[StandardModesLog.txt](https://github.com/user-attachments/files/21729613/StandardModesLog.txt)

DonLakeFlyer, 2025-08-12:
Do you have gstreamer debug logs turned on? It's super hard to look at this with all the GST stuff. If so could you turn that off?

DonLakeFlyer, 2025-08-31:
Ah, ok. Not sure why this is still in "development" but nothing to do from the QGC side of things.