mavlink/qgroundcontrol #13312: QGroundControl 5 stable release breaks

covelus, 2025-08-17:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
QGC opened when screen get unlocked. Mission keeps being executed in px4_sitl. 

**Current Behavior**
Possible computer suspension, but it breaks, and produces an error in PX4, aborting mission, and going into return mode., Still, while QGC breaks, px4_sitl and Gazebo, even while paused, resume working, but the simulation goes into return mode. 

**Steps To Reproduce**
1. Tested in GNU/Linux, open QGC and PX4 simulation, e.g: make px4_sitl gz_x500
2. Run a mid or long mission, like multi-wayponts (8 km in my case). Let it running until the screen gets locked.
3. Wait about 10 or 15 minutes, unlock screen and find if still running.

**Additional Details**
check logs below

### System Information

QGC Version: 5.0.6 64 bit 
QGC build: stable
Operating System: Ubuntu 24.04
Flight Controller: px4_sitl
Autopilot Software (with version): customized PX4, latest GH version (QGC displays 1.16.0rc)

### Log Files and Screenshots

In px4_sitl: 
[commander] Connection to ground station lost	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
WARN  [failsafe] Failsafe activated: entering Hold for 5 seconds	

Part of the logs when it run the mission (running ./QGroundControl.AppImage --logging:full )

    12.470 - debug: void Vehicle::_sendMavCommandWorker(bool, bool, const MavCmdAckHandlerInfo_t*, int, MAV_CMD, MAV_FRAME, float, float, float, float, double, double, float) command:param1-7 512 280 0 0 0 0 0 0 (VehicleLog:unknown:0)
    12.470 - debug: void Vehicle::_sendMavCommandFromList(int) command:tryCount:param1-7 "MAV_CMD_REQUEST_MESSAGE" 0 280 0 0 0 0 0 0 (VehicleLog:unknown:0)
    12.473 - debug: "_handleCommandAck command(MAV_CMD_REQUEST_MESSAGE) result(MAV_RESULT_DENIED)" (VehicleLog:unknown:0)
    14.055 - debug: Bounding cube: QGeoCoordinate(47.4306149, 8.5461639, 25) QGeoCoordinate(47.3979709, 8.6419602, 25) (MissionControllerLog:unknown:0)
   37.175 - debug: "_handleCommandAck command(MAV_CMD_DO_SET_MODE) result(MAV_RESULT_ACCEPTED)" (VehicleLog:unknown:0)
    37.175 - debug: _handleCommandAck Ack not in list "MAV_CMD_DO_SET_MODE" (VehicleLog:unknown:0)
    37.816 - debug: First call to begindResetModel - calling QAbstractListModel::beginResetModel QmlObjectListModel(0x614c24e64b40) (QmlObjectListModelLog:unknown:0)
    37.816 - debug: _resetModelNestingCount: 1 QmlObjectListModel(0x614c24e64b40) (QmlObjectListModelLog:unknown:0)
    37.816 - debug: _resetModelNestingCount: 0 QmlObjectListModel(0x614c24e64b40) (QmlObjectListModelLog:unknown:0)
    37.816 - debug: Last call to endResetModel - calling QAbstractListModel::endResetModel QmlObjectListModel(0x614c24e64b40) (QmlObjectListModelLog:unknown:0)
    37.817 - debug: _handleHeartbeat updating lastCurrentIndex from cached value: 7 (MissionManagerLog:unknown:0)
    37.817 - debug: void Vehicle::_sendMavCommandWorker(bool, bool, const MavCmdAckHandlerInfo_t*, int, MAV_CMD, MAV_FRAME, float, float, float, float, double, double, float) command:param1-7 400 1 0 0 0 0 0 0 (VehicleLog:unknown:0)
    37.817 - debug: void Vehicle::_sendMavCommandFromList(int) command:tryCount:param1-7 "MAV_CMD_COMPONENT_ARM_DISARM" 0 1 0 0 0 0 0 0 (VehicleLog:unknown:0)
    37.917 - debug: "_handleCommandAck command(MAV_CMD_COMPONENT_ARM_DISARM) result(MAV_RESULT_ACCEPTED)" (VehicleLog:unknown:0)
    37.917 - debug: Dropping message (expected as event): "Armed by external command\t" (VehicleLog:unknown:0)
    37.918 - debug: Incoming event: last seq=21, msg seq=22, id=0190caf3 (EventsLog:unknown:0)
    37.918 - debug: Got Event: ID: 26266355 namespace: px4 name: commander_armed_by msg: Armed by external command (EventsLog:unknown:0)
    37.918 - debug: Incoming event: last seq=22, msg seq=23, id=01ee01b1 (EventsLog:unknown:0)
    37.918 - debug: Got Event: ID: 32375217 namespace: px4 name: mission_execute msg: Executing Mission (EventsLog:unknown:0)
    37.918 - debug: Incoming event: last seq=23, msg seq=24, id=017d425c (EventsLog:unknown:0)
    37.918 - debug: Got Event: ID: 24986204 namespace: px4 name: mission_climb_before_start msg: Climb to 25.0 m above home (EventsLog:unknown:0)
    37.918 - debug: Dropping message (expected as event): "Executing Mission\t" (VehicleLog:unknown:0)
    38.020 - debug: Dropping message (expected as event): "Climb to 25.0 meters above home\t" (VehicleLog:unknown:0)
    38.774 - debug: First call to begindResetModel - calling QAbstractListModel::beginResetModel QmlObjectListModel(0x614c24cd9e88) (QmlObjectListModelLog:unknown:0)
    38.774 - debug: _resetModelNestingCount: 1 QmlObjectListModel(0x614c24cd9e88) (QmlObjectListModelLog:unknown:0)
    38.774 - debug: _resetModelNestingCount: 0 QmlObjectListModel(0x614c24cd9e88) (QmlObjectListModelLog:unknown:0)
    38.774 - debug: Last call to endResetModel - calling QAbstractListModel::endResetModel QmlObjectListModel(0x614c24cd9e88) (QmlObjectListModelLog:unknown:0)
    39.597 - debug: Dropping message (expected as event): "Takeoff detected\t" (VehicleLog:unknown:0)
    39.597 - debug: Incoming event: last seq=24, msg seq=25, id=01414569 (EventsLog:unknown:0)
    39.597 - debug: Got Event: ID: 21054825 namespace: px4 name: commander_takeoff_detected msg: Takeoff detected (EventsLog:unknown:0)
  1693.057 - debug: _updateMissionIndex currentIndex: 8 (MissionManagerLog:unknown:0)
  1693.058 - debug: _updateMissionIndex caching _lastCurrentIndex for possible update: 8 (MissionManagerLog:unknown:0)
  1693.058 - debug: Dropping message (expected as event): "Mission finished, loitering\t" (VehicleLog:unknown:0)
  1693.058 - debug: Incoming event: last seq=25, msg seq=26, id=014762f6 (EventsLog:unknown:0)
  1693.058 - debug: Got Event: ID: 21455606 namespace: px4 name: mission_finished_loiter msg: Mission finished, loitering (EventsLog:unknown:0)
  1693.058 - debug: Incoming event: last seq=26, msg seq=27, id=01352ce4 (EventsLog:unknown:0)
  1693.058 - debug: Got Event: ID: 20262116 namespace: px4 name: commander_rtl msg: Returning to launch (EventsLog:unknown:0)
  1693.058 - debug: Incoming event: last seq=27, msg seq=28, id=01522694 (EventsLog:unknown:0)
  1693.058 - debug: Got Event: ID: 22161044 namespace: px4 name: vrtl_return_at msg: RTL: start return at 31 m (30 m above destination) (EventsLog:unknown:0)
  1693.058 - debug: Dropping message (expected as event): "Returning to launch\t" (VehicleLog:unknown:0)
  1693.059 - debug: Dropping message (expected as event): "RTL: start return at 31 m (30 m above destination)\t" (VehicleLog:unknown:0)
  1693.064 - debug: First call to begindResetModel - calling QAbstractListModel::beginResetModel QmlObjectListModel(0x614c24e64b40) (QmlObjectListModelLog:unknown:0)
  1693.064 - debug: _resetModelNestingCount: 1 QmlObjectListModel(0x614c24e64b40) (QmlObjectListModelLog:unknown:0)
  1693.064 - debug: _resetModelNestingCount: 0 QmlObjectListModel(0x614c24e64b40) (QmlObjectListModelLog:unknown:0)
  1693.064 - debug: Last call to endResetModel - calling QAbstractListModel::endResetModel QmlObjectListModel(0x614c24e64b40) (QmlObjectListModelLog:unknown:0)
Bus error (core dumped)

covelus, 2025-08-20:
It is true that OS suspending apps (while suspending itself) happen, true, but an app should **not** break because the OS suspends it, this just shows an unacceptable lack or software robustness. 
Moreover, I have recently seen QGC also breaking while using it... I will try to reproduce it, collect more logs, and upload them here.

DonLakeFlyer, 2025-08-20:
I missed the fact that by "break" you mean "crash".

DonLakeFlyer, 2025-09-01:
@covelus Plan file?

covelus, 2025-09-03:
I'm 100% sure the plan file has nothing to do at all with the QGC breaking... 
It just happened to me twice today, in GNU/Linux and MacOS. 
In MacOS was where I was loading some param file to restore those on a drone.  
I loaded the file for review, and once clicked OK, it broke as well, and the params file I'm 100% sure is good. 

I had run it from the terminal to replicate it, so, I got the shell log, and also MacOS log: 

[MacOS.log](https://github.com/user-attachments/files/22126150/MacOS.log)
[Terminal.log](https://github.com/user-attachments/files/22126151/Terminal.log)

covelus, 2025-09-04:
OK, here it is
[mission-move-example_short_zurich25m.plan.txt](https://github.com/user-attachments/files/22145126/mission-move-example_short_zurich25m.plan.txt)

covelus, 2025-09-22:
I upgraded to the last version, 5.0.7, and it keeps breaking. 
Every time I connect flight controller (and I tried with 2 Pixhawk 6c), change a parameter, and click on restart, it breaks, also happens sometimes when I disconnect the USB wire

... Leaving terminal's --logging:full logs, I hope it helps:

[2025-09-21-QGroundControl_break_when_changing_parameter.log](https://github.com/user-attachments/files/22464500/2025-09-21-QGroundControl_break_when_changing_parameter.log)
[2025-09-21-QGroundControl_break_when_disconnecting.log](https://github.com/user-attachments/files/22464503/2025-09-21-QGroundControl_break_when_disconnecting.log)
[2025-09-22-QGroundControl_break_when_changing_parameter.log](https://github.com/user-attachments/files/22464501/2025-09-22-QGroundControl_break_when_changing_parameter.log)

Thank you

DonLakeFlyer, 2025-09-22:
Also can you get me a stack trace like you did in the previous MacOS.log for the parameter change crashes?

covelus, 2025-09-22:
> [@covelus](https://github.com/covelus) In the cases where it crashes does the vehicle have a mission already loaded onto it?

True, I forgot that detail, good question: 
No, it had no plan on it, it was empty... Indeed, if I reboot the drone without even changing any parameters, it also breaks. 
And I tried with different devices (2 models): 

- Holybro Pixhawk 6C alu (attached to a X500 V2)
- Holybro Pixhawk 6C plastic, nothing attached to it
- Holybro Pixhawk 6X, just an I2C device attached to it, not even completely configured

> Also can you get me a stack trace like you did in the previous MacOS.log for the parameter change crashes?

OK. 
The previous were Linux logs, after you request, I switched to a MacOS, I tested 5.0.6 stable, it did not break, I upgraded to, the latest stable release, 5.0.7, I repeated the test (just restart the Pixhawk 6X) and I confirm it breaks as in Linux, so I'm attaching below the MacOS Problem Report (~52 KB), and also the Terminal logs (~10.1 MB):

_If you need me to do something else, please, just let me know. Although I do not have the QGC development stack configured, I hope you can reproduce the issue. Alternatively, I could try to find the time to install it, and we could organise a remote call to debug if you want._

Thank you and best regards

DonLakeFlyer, 2025-09-22:
Once #13417 is done can you pull an artifact out of the GitHub actions and give it a try?