mavlink/qgroundcontrol #13325: "Return" and "Land" button malfunction because "Unknown flight Mode"

foreverlz1111, 2025-08-21:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
Hello, I am running with the latest QGC source code. I tried compiling and running from a clean source, and tested each basic function **if it works well**, such as "Takeoff", "Land", "Return", and "Go to Here" button in Homepage.

**Current Behavior**
No reaction when click "Land" or "Return" button in QGC software

**Steps To Reproduce**
1. Run PX4 simulator ` make px4_sitl jmavsim`
2. Run QGC software. `./build/Debug/QGroundControl`
3. Click "Takeoff" button in QGC software.
4. Click "Land" or "Return" button in QGC software.

**Additional Details**
I used debug found out "./src/FirmwarePlugin/PX4/PX4FirmwarePlugin.cc" line 316 and line 321 cause this error.

### System Information

QGC Version: **5.0.6**
QGC build: **stable**
Operating System: **Ubuntu 25.04**
Flight Controller: None
Autopilot Software (with version): **PX4 1.16**

### Log Files and Screenshots

_**Screenshot**_
[image]

**_QGC console log :_** 
`default::     76.797 - warning: Unknown flight Mode "Return" (PX4FirmwarePlugin::setFlightMode:150)
VehicleLog::     76.797 - warning: FirmwarePlugin::setFlightMode failed, flightMode: "Return" 
(Vehicle::setFlightMode:1688)

default::     78.189 - warning: Unknown flight Mode "Return" (PX4FirmwarePlugin::setFlightMode:150)
VehicleLog::     78.189 - warning: FirmwarePlugin::setFlightMode failed, flightMode: "Return" (Vehicle::setFlightMode:1688)

default::     79.554 - warning: Unknown flight Mode "Return" (PX4FirmwarePlugin::setFlightMode:150)
VehicleLog::     79.554 - warning: FirmwarePlugin::setFlightMode failed, flightMode: "Return" (Vehicle::setFlightMode:1688)
`

**_My modified, then it works well.:_**
`
// _begin at "./src/FirmwarePlugin/PX4/PX4FirmwarePlugin.cc" line 313_
void PX4FirmwarePlugin::guidedModeRTL(Vehicle* vehicle, bool smartRTL) const
{
    Q_UNUSED(smartRTL);
\-     _setFlightModeAndValidate(vehicle, _rtlFlightMode);
\+    _setFlightModeAndValidate(vehicle, rtlFlightMode());
}

void PX4FirmwarePlugin::guidedModeLand(Vehicle* vehicle) const
{
\-     _setFlightModeAndValidate(vehicle, _landingFlightMode);
\+    _setFlightModeAndValidate(vehicle, landFlightMode());
}
`

dakejahl, 2025-09-02:
https://github.com/mavlink/mavlink/pull/2191 strikes again

julianoes, 2025-09-02:
@foreverlz1111 can you just confirm which git hash of PX4 you're using to test this?

julianoes, 2025-09-03:
Ok, thanks. So your 1.16 is really main, but that's fine, and does sound like a bug somewhere indeed.

closed 2025-09-05 by dakejahl

foreverlz1111, 2025-09-05:
> [@foreverlz1111](https://github.com/foreverlz1111) Can you test my fix linked here from the artifact in the pull?

Latest commit works well, QGC sending correct commands to PX4 now.😃😃
`commit f7bd62f489c19671f20891af9df7c61273bb522e`

[image]

```
INFO  [navigator] RTL: start return at 53 m (3 m above destination)
INFO  [navigator] RTL: land at destination
INFO  [commander] Landing detected
INFO  [commander] Disarmed by landing
INFO  [tone_alarm] notify neutral
INFO  [logger] closed logfile, bytes written: 11192395
INFO  [commander] Armed by external command
INFO  [tone_alarm] arming warning
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2025-09-05/01_48_22.ulg
INFO  [logger] Opened full log file: ./log/2025-09-05/01_48_22.ulg
INFO  [commander] Takeoff detected
INFO  [navigator] RTL: start return at 77 m (30 m above destination)
INFO  [navigator] RTL: land at destination
INFO  [commander] Landing detected
INFO  [commander] Disarmed by landing

```

PR #13352 (2025-09-05): PX4: Fix bugs with flight mode names changing from Standard Modes
* Fix #13325
* This is just a quick fix. In reality the whole bag of code surrounding standard modes and flight modes needs to be re-written. I know what I want to do. I'll deal with that in master.

@dakejahl Can you review and test? I did minimal testing myself but ran out of time till tomorrow. I'll poke at it some more then. I think there is still ArduPilot side problems as well.