mavlink/qgroundcontrol #13334: RTSP video stream from VOXL2 disconnects after a few seconds in QGroundControl

damiananaalexia, 2025-08-25:
Hello,

I am streaming video from a ModalAI VOXL2 using voxl-streamer into QGroundControl. The stream initially appears, but after a few seconds it disconnects and QGC shows “Waiting for video” again.

I tested with different resolutions and bitrates (640x480, 1024x768, 1 Mbps, 3 Mbps, etc.). The issue happens consistently: the video appears for a short time, then drops.

From the VOXL2 side (tail -f /var/log/syslog | grep voxl-streamer), I see that QGC connects and disconnects immediately:
-----------------------------------------------------------------------------
A new client has connected, total clients: 1
rtsp client disconnected, total clients: 0
no more rtsp clients, closing source pipe intentionally
---------------------------------------------------------------------------
So it seems QGC connects to the RTSP stream but does not keep it alive.

Versions
	•	Ubuntu: 24.04.1 LTS (noble)
	•	VOXL Suite / Services:
	•	voxl-camera-server: 0.5.9
	•	voxl-streamer: 0.7.1
	•	voxl-vision-hub: 1.6.6
	•	QGroundControl: v5.0.6.64 (64-bit)
I tried both H264 and H265, and different resolutions (640x480, 1024x768), but the issue persists. Is there a recommended configuration (codec, bitrate, resolution) for stable video streaming to QGC?

HTRamsey, 2025-08-31:
Does it display in something like VLC reliably? If so, next step would be to enable GStreamer logging and see if there's anything obvious from that

HTRamsey, 2025-09-02:
Make sure you are using the VLC from Snap rather than Apt to test it. The one from Apt no longer includes the RTSP support