mavlink/qgroundcontrol #13363: Breaking change with camera protocol v2 discovery in QGC v5

superware, 2025-09-07:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
Discover camera, play video stream and show UI related to `CAMERA_CAP_FLAGS_HAS_TRACKING_POINT`.

**Current Behavior**
Camera only sometimes being discovered, the "tracking point" UI button never shows up.

**Steps To Reproduce**
1. A companion computer is sending `HEARTBEAT` as `MAV_COMP_ID_CAMERA` with type `MAV_TYPE_CAMERA`.
2. Per QGC request, send `CAMERA_INFORMATION` with flags `CAMERA_CAP_FLAGS_HAS_VIDEO_STREAM | CAMERA_CAP_FLAGS_HAS_TRACKING_POINT | CAMERA_CAP_FLAGS_HAS_TRACKING_RECTANGLE`.
3. Send `CAMERA_SETTINGS` and `VIDEO_STREAM_INFORMATION` pre request, `VIDEO_STREAM_STATUS` continuously.
4. Tracking-point UI is never shown.
5. If QGC is launched first and "gives up" on detecting gimbals/cameras, then launching the companion computer doesn't get the camera discovered (only `GIMBAL_MANAGER_INFORMATION` is being requested).
6. If Application Settings -> Video -> Video Source is set to "Video Stream Disabled" (thinking this should mean "there's no default video source, only discovered at runtime", but it seems the first video stream discovered is changing this "default" video source), not sure this is directly related.

**Additional Details**
It seems that the discovery logic described in MAVLink [Camera Protocol v2](https://mavlink.io/en/services/camera.html#camera-discovery) isn't strictly followed.

### System Information

QGC Version: 5.0.6 (works in 4.4.5)
QGC build: stable
Operating System: Windows 10
Flight Controller: CubePilot Cube Orange
Autopilot Software: PX4 1.15.4

Thanks!

DonLakeFlyer, 2025-09-15:
@julianoes Can you help out here?

SudoPotato1, 2025-10-16:
@DonLakeFlyer [20251016 QGC video log.txt](https://github.com/user-attachments/files/22942524/20251016.QGC.video.log.txt)
Did not work.