mavlink/qgroundcontrol #13367: Arm64 linux appimage

jack-obrien, 2025-09-09:
### Feature Description

Hi there, I am working on QGC for the raspberry pi. I am able to compile it and run it like a normal cpp project in linux. Would you be interested in me working on the necessary changes to cmake scripts and CI, with the end goal being a linux arm64 appimage added to the usual releases?

No promise on the timeline for this work, it might take several months to finish with some breaks in between contributions as life gets busy.

Please let me know if this sounds good! I have been using this app for over a year and want to help improve the project and make it accessible to more users.

HTRamsey, 2025-09-13:
This already exists in the daily, just never made it to stable I don't think

jack-obrien, 2025-09-16:
Oh wow, nice work!

closed 2025-09-16 by jack-obrien

jack-obrien, 2025-09-16:
Hey, just tried it out on my CM5 running the latest Raspberrry Pi OS Bookworm. It gave an Exec format error - this is actually the same issue I ran into when compiling QGC for myself.

I can provide more details later when I get time, but I believe the issue is that the prebuilt Qt 6.8.3 binaries provided by Qt are built on Ubuntu 24.04 which has quite a new version of glibc. Those .so files link against some libc functionality which is not available on Raspberry Pi OS Bookworm (AKA debian 12). 

So this issue should be fixed if you use a relatively new OS on your Raspberry Pi. But there is no official Raspberry Pi OS yet for Debian 13.

Not sure if this issue is also present on x86_64 debian 12. It depends on the build platform used by Qt, I guess.

Probably don't worry about fixing this, as it would require compiling your own Qt 6.8.3 to work on Raspberry Pi OS bookworm, which adds a little complexity, and this problem will probably go away soon.

HTRamsey, 2025-09-16:
The oldest github runner for ubuntu arm is 24.04, so yeah that's probably your problem

DonLakeFlyer, 2025-09-16:
Qt notes for 6.8 on arm:
"Note: For Linux on Arm on desktops, we use Raspberry Pi 5 with 8GB RAM and Ubuntu 24.04 as a reference platform. If you have an issue with another hardware, please try to reproduce the issue on Raspberry Pi 5. We plan to move to regular platform support when a wider range of ARM-based desktop hardware becomes available. The official binaries are built on Ubuntu 24.04 which ships with glibc 2.39, if your glibc is older than what we build with, you need to rebuild from sources."

https://doc.qt.io/qt-6.8/supported-platforms.html#linux-on-arm-desktop-support

soboln9, 2025-09-22:
I want to create .AppImage!

soboln9, 2025-09-22:
Thank you very much, everything worked out perfectly!!!