mavlink/qgroundcontrol #13380: Windows Installation

balasuresh19, 2025-09-13:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
i have to run qgroundcontrol in windows
**Current Behavior**
unable to configure qgc it showing error that pkgconfig.exe is not availble

**Steps To Reproduce**
1.
2.
3.

**Additional Details**
Add any extra info (errors, warnings, etc.) here.

### System Information

in linux i installed and run qgc but unable to install in windows it gives error .its possible to get  windows file in linux

HTRamsey, 2025-09-15:
Does that file actually exist in your gstreamer install?

HTRamsey, 2025-09-15:
First, pkgconfig is provided by a GStreamer install so make sure you have 1.22 installed in the default location correctly. And no you can't build a windows exe on linux

HTRamsey, 2025-09-16:
https://gstreamer.freedesktop.org/data/pkg/windows/1.22.12/msvc/gstreamer-1.0-devel-msvc-x86_64-1.22.12.msi
https://gstreamer.freedesktop.org/data/pkg/windows/1.22.12/msvc/gstreamer-1.0-msvc-x86_64-1.22.12.msi

Probably soon I will try to make it such that you don't have to have GStreamer installed, there will be a CMake fallback for it