mavlink/qgroundcontrol #13402: RTK base detected in Mission Planner but not showing RTK icon in QGroundControl

ZoneDK-sudo, 2025-09-17:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
When RTK GPS is plugged in, QGC should detect it and display the RTK icon in the top status bar.

**Current Behavior**
Mission Planner correctly detects RTK, shows satellites and solution.
Windows Device Manager shows the COM ports and sensors.
QGC does not show the RTK icon when connected to the same hardware (latest release on Windows).

**Steps To Reproduce**
Connect RTK GPS module to laptop (Windows).
Open Mission Planner → verify RTK detected and working.
Close Mission Planner
Open QGroundControl (latest release) with the same setup.
Observe that no RTK icon appears.

**Additional Details**
Device Manager confirms COM port is present.
RTK receiver is functioning as expected with other ground stations.
Only QGC fails to show icon.

### System Information

QGC Version: Latest official release (tested on [5.0.7])
QGC Build: Stable release for Windows
Operating System: Windows 11
Flight Controller: [Pixhawk 6c mini]
Autopilot Software: [PX4 v1.16]

### Log Files and Screenshots

https://review.px4.io/plot_app?log=51e97040-7704-4929-8e23-ea5a924d96c4

HTRamsey, 2025-09-17:
I think you have to have an active mavlink connection for the icon to show

DonLakeFlyer, 2025-09-17:
* Boot QGC without RTK plugged in
* Turn on LinkManagerLog and LinkManagerVerboseLog console logging: https://docs.qgroundcontrol.com/Stable_V5.0/en/qgc-user-guide/settings_view/console_logging.html
* [image]

* Plug in RTK
* Save console log to file and post here

ZoneDK-sudo, 2025-09-17:
unplugged rtk setting the options 

[image]

DonLakeFlyer, 2025-09-17:
The problem is the fact that the RTK devices comes through on a "Virtual com port" device. Which in turn has no vendor/product id info:
```
  183.835 - debug: portName:           "COM9" (qgc.comms.linkmanager:verbose:unknown:0)
   183.836 - debug: systemLocation:     "\\\\.\\COM9" (qgc.comms.linkmanager:verbose:unknown:0)
   183.836 - debug: description:        "u-blox Virtual COM Port" (qgc.comms.linkmanager:verbose:unknown:0)
   183.837 - debug: manufacturer:       "u-blox AG" (qgc.comms.linkmanager:verbose:unknown:0)
   183.837 - debug: serialNumber:       "" (qgc.comms.linkmanager:verbose:unknown:0)
   183.837 - debug: vendorIdentifier:   0 (qgc.comms.linkmanager:verbose:unknown:0)
   183.837 - debug: productIdentifier:  0 (qgc.comms.linkmanager:verbose:unknown:0)
```

This doesn't allow us to know that it's an RTK we support. Not sure how to fix this off the top of my head. Will take some researching.

DonLakeFlyer, 2025-09-17:
> I think you have to have an active mavlink connection for the icon to show

Not for RTK GPS

ZoneDK-sudo, 2025-09-18:
on different windows 11 computer 

[image]

ZoneDK-sudo, 2025-09-18:
Problem PC
[image]

Working PC
[image]

ZoneDK-sudo, 2025-09-18:
[tes2.txt](https://github.com/user-attachments/files/22413093/tes2.txt)

ZoneDK-sudo, 2025-09-18:
also why dont I get GPS RTK Lock?

[image]

closed 2025-09-20 by DonLakeFlyer