mavlink/qgroundcontrol #13435: Commlink UDP target is not always getting added.

Edmonds1997, 2025-09-25:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
Describe what you thought should happen here.
I expect all three UDP comm links to work and always show the three vehicles I am using.

**Current Behavior**
I am using QGroundControl in conjunction with PX4 SITL and AirSim/Unreal Engine. The only way I can reliably get all three of the vehicles to show up in QGC is to follow the exact steps in the "Steps To Reproduce" section.  If I try to connect the 3 commlinks after the PX4 instances are running, only vehicles 2 and 3 will show up.

[image]

**Steps To Reproduce**
1.)Launch QGC and connect all 3 UDP comm links
2.)Start AirSim simulator
3.)Start PX4 instances (at this point the vehicles still arent showing up in GQC)
4.)Disconnect all 3 comm links
5.)Reconnect all 3 comm links (now all 3 vehicles show up fine)

[image]

**Additional Details**
CONSOLE LOGS - 

I dont see log messages for debug: UDP Adding target: QHostAddress("192.168.2.166") 14580       OR
debug: _addLink" "Drone1" If I connect the comm links after the PX4 instances are running.

[All_3_working.txt](https://github.com/user-attachments/files/22536912/All_3_working.txt)

[Only_2_working.txt](https://github.com/user-attachments/files/22536927/Only_2_working.txt)

### System Information

QGC ver 5.0.7 64 bit
installed from windows installer
Windows 11
PX4 Sitl with AirSim/Unreal Engine

DonLakeFlyer, 2025-09-25:
Sure

Edmonds1997, 2025-10-01:
I have tried a couple of things.

1.) Tried my same setup but with QGC running on Linux. It seems to connect much more reliably and without the need to disconnect and reconnect on Linux.

2.) Tried running without airsim. Again linux seems much more reliable and without the need to disconnect and reconnect. On windows sometimes it wont connect at all and sometimes I have to disconnect and reconnect the comm link for the vehicle to actually show up.

DonLakeFlyer, 2026-03-14:
@Edmonds1997 ?